WPIRoboticsProjects/GRIP #795: Running GRIP on RPI3

BradHacker, 2017-01-21:
We are trying to run GRIP generated code on our raspberry pi 3. We have connected a webcam to our PC and used the Windows GRIP Computer Vision Engine software to adjust our filters and output some contours. We have generated the code in a python file and we are using another python file to access the GripPipeline. We are then publishing the x and y values of the contours to our NetworkTables. The problem lies in the fact that we are not getting any output whatsoever. We have a second webcam running at the same time that does output the contours in the PC software, but the pi outputs nothing. We are running the NetworkTables successfully. We have the rpi connected to a router that the roboRio is also connected to. We have input a test line into the NetworkTables and it showed up in our drivers station. The problem lies with the grip code. We can ping the roboRio from the pi and we can ping the pi from the roboRio. We just recently changed the values in our hsl filter and generated the code and it no longer works. We then changed the values manually in the working grip file and it worked. So the code in the GRIP program works fine but when it is generated and transferred to the pi, it does not produce any contours. If we edit the code on the pi and change the hsl filter to something with a much wider range, it works fine and outputs a bunch of contours in networktables. So something between GRIP on our computer and GRIP/OpenCV on the pi is inconsistent. We aren't sure what is going on.

bradamiller, 2017-01-22:
It's possible that there is a bug in GRIP that caused it to generate an incorrect Python program. Do you know Python well enough to debug the program? Perhaps you could take the PI off the robot and use a keyboard, mouse, and monitor to run the program. Then you could just use print statements to see where it's broken.

If hyou're not sure how to do that could you post the grip save file and maybe we could take a look to see if there's something obviously wrong?

brookshank, 2017-01-26:
I am having the same issue, where a wider range of values works fine but when the pipeline is set to find only a small range in contours then no contours are actually found. http://pastebin.com/YFu0wgp4 this is my grip file, and http://pastebin.com/X0edmjwT is my python code for running the grip pipeline.

brookshank, 2017-02-06:
#806 I posted about it too, we're not having much luck over there. Have you found a solution?