WPIRoboticsProjects/GRIP #796: GRIP crashes when closing a running operation.

AmirMenad, 2017-01-21:


SamCarlberg, 2017-01-23:
Please be more specific. What operation was running? Did you close the app window, or the X button on the operation in the pipeilne?

AmirMenad, 2017-01-24:
By clicking the X button on the operation while it's running, it will throw a java error and freeze up.

SamCarlberg, 2017-01-24:
Can you post the error? And what operation(s) does this happen with?

JLLeitschuh, 2017-02-07:
This is probably a deadlock between the pipeline thread and the java FX thread.

I'm not really certain what thread condition would cause a deadlock but it probably is a convergence between the JavaFx thread and the pipeline runner thread.

Maybe a quick fix would be to run the removal of the step from the pipeline in a different thread than the javafx thread.