WPIRoboticsProjects/GRIP #802: Getting This Error When Trying to Run GRIP on Raspberry Pi 3 B.

Jetlogan, 2017-01-27:
pi@raspberrypi:~ $ env LD_LIBRARY_PATH=/home/pi/vision/grip:$LD_LIBRARY_PATH java -jar /home/pi/vision/grip/grip.jar /home/pi/vision/grip/project.grip
Jan 27, 2017 12:58:49 AM java.util.logging.LogManager$RootLogger log
CONFIG: Configuration done.
Jan 27, 2017 12:58:49 AM java.util.logging.LogManager$RootLogger log
CONFIG: GRIP Version: v1.5.1
2017-01-27 00:58:51.879:INFO::main: Logging initialized @2802ms
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x766d46f4, pid=1305, tid=1983337568
#
# JRE version: Java(TM) SE Runtime Environment (8.0_65-b17) (build 1.8.0_65-b17)
# Java VM: Java HotSpot(TM) Client VM (25.65-b01 mixed mode linux-arm )
# Problematic frame:
# V  [libjvm.so+0x2e36f4]
#
# Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /home/pi/hs_err_pid1305.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
#
Aborted

Been getting this when I try to run the GRIP program. Re-installed Java and tried Java 7. Ask if you need the full error file.

SamCarlberg, 2017-01-27:
A couple things to note:

1. GRIP requires Java 8
2. Running GRIP from the command line changed in 1.5.0. [See this wiki page](https://github.com/WPIRoboticsProjects/GRIP/wiki/Command-line-options)
3. We don't recommend running GRIP directly on a pi. You should be generating code and using it in a program. There are examples [here](https://www.github.com/WPIRoboticsProjects/GRIP-code-generation)

Post the hs_err log and I'll see if it's a problem on our end.

nightpool, 2017-01-27:
@SamCarlberg This looks like it's using java 8 to me? `JRE version: Java(TM) SE Runtime Environment (8.0_65-b17) (build 1.8.0_65-b17)`

JLLeitschuh, 2017-01-27:
If you really want to run it on a PI this exists but is not officially supported:
https://github.com/WPIRoboticsProjects/GRIP/wiki/Running-GRIP-on-a-Raspberry-Pi-2

Jetlogan, 2017-01-28:
@SamCarlberg Yeah I'm using Java 8 on the Pi. I'll look at those wiki links.

Jetlogan, 2017-01-28:
Here is the hs_err

```
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x766d46f4, pid=1305, tid=1983337568
#
# JRE version: Java(TM) SE Runtime Environment (8.0_65-b17) (build 1.8.0_65-b17)
# Java VM: Java HotSpot(TM) Client VM (25.65-b01 mixed mode linux-arm )
# Problematic frame:
# V  [libjvm.so+0x2e36f4]
#
# Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
#

---------------  T H R E A D  ---------------

Current thread (0x76207000):  JavaThread "main" [_thread_in_vm, id=1306, stack(0x76326000,0x76376000)]

siginfo: si_signo: 11 (SIGSEGV), si_code: 1 (SEGV_MAPERR), si_addr: 0x0000118c

Registers:
  r0  = 0x00000005
  r1  = 0x00000006
  r2  = 0x00000000
  r3  = 0x00000678
  r4  = 0x76207000
  r5  = 0x7698efb8
  r6  = 0x0000118c
  r7  = 0x762059fc
  r8  = 0x00000000
  r9  = 0x63831140
  r10 = 0x76207000
  fp  = 0x76373ec4
  r12 = 0x00000004
  sp  = 0x76373e80
  lr  = 0x766d46d0
  pc  = 0x766d46f4
  cpsr = 0x60000010

Top of Stack: (sp=0x76373e80)
0x76373e80:   76373ebc 76207000 00000000 762075f0
0x76373e90:   7698efb8 76373f03 63831140 76207000
0x76373ea0:   63831140 76205888 76373f03 7620713c
0x76373eb0:   0000118c 00000000 762059fc 76373f5c
0x76373ec0:   76373f54 640d97bc 7416ec40 640d9798
0x76373ed0:   00000000 00000000 76373f5c 7413d3fc
0x76373ee0:   76207000 76373efc 762059fc 76207000
0x76373ef0:   76373f10 76373f08 76373f2c 7668d96c 

Instructions: (pc=0x766d46f4)
0x766d46d4:   e3a03000 e1520003 e50b4040 e50b303c
0x766d46e4:   0a000001 e24b0040 eb051471 e59f30ac
0x766d46f4:   e5966000 e7953003 e5933000 e7963003
0x766d4704:   e3530000 0a000021 e1a00006 ebff0565 

Register to memory mapping:

  r0  = 0x00000005
0x00000005 is an unknown value

  r1  = 0x00000006
0x00000006 is an unknown value

  r2  = 0x00000000
0x00000000 is an unknown value
[... 532 lines cut ...]
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2709
Revision	: a22082
Serial		: 0000000082547311

Memory: 4k page, physical 947732k(429228k free), swap 102396k(102396k free)

vm_info: Java HotSpot(TM) Client VM (25.65-b01) for linux-arm-vfp-hflt JRE (1.8.0_65-b17), built on Oct  6 2015 16:19:04 by "java_re" with gcc 4.7.2 20120910 (prerelease)

time: Fri Jan 27 00:58:53 2017
elapsed time: 4 seconds (0d 0h 0m 4s)
```

SamCarlberg, 2017-01-28:
That's a problem with network tables. Not our code causing the issue

Jetlogan, 2017-01-28:
Then how would I get this to work? And also this isn't a unsupported OS because I am using the OS you specified in your Raspberry Pi Tutorial.

bradamiller, 2017-01-28:
I just forwarded the request to one of the developers who has been working on the libraries for the Raspberry PI. Where did you get the libraries that you were using, and what version of Linux are you running on the PI?  It would be helpful if you can post the smallest program that fails to help reproduce/debug the issue.

bradamiller, 2017-01-28:
Also make sure that you are using the arm-raspbian builds and not the armhf builds. I think that would cause that crash.

Jetlogan, 2017-01-28:
@bradamiller I got all of my files from this tutorial: https://github.com/WPIRoboticsProjects/GRIP/wiki/Running-GRIP-on-a-Raspberry-Pi-2

I wouldn't think the two different pis would make a difference as they are both running the same OS.

defied, 2017-04-03:
Any progress on this?