WPIRoboticsProjects/GRIP #803: Publish Video only appears to send to localhost

WhyGitHub1, 2017-01-27:
Running Grip on Kangaroo.  Publishing Video and Network table contours to Driver Station laptop at roboRIO-xxx-FRC.local.  The Network tables work as expected.  However am unable to get Video on Smartdashboard at driver station using the sdb-grip Grip viewer extension.  
Note that running Grip and Smartdashboard on the Kangaroo with this extension does show the camera video with Grip set for either localhost or roboRIO-xxx-FRC.local.   Its as thought the publish video stays at localhost.

JLLeitschuh, 2017-01-27:
So the video server is running on the kangaroo. You need to figure out what the IP address of the kangaroo device is and use that to connect to it there.

@SamCarlberg Do we need to update the smart dashboard plugin to support GRIP running on devices other than at the IP: `roboRIO-xxx-FRC.local`?

SamCarlberg, 2017-01-27:
I'm not familiar with the SD plugin.

WhyGitHub1, 2017-01-27:
Grip Smartdashboard Extension.  On GitHub at WPIRobotisProjects/GRIP-SmartDashboard.   Need for this may be OBE if you change to the stream format used by the dashboard viewer.
I'll try the Kangaroo's address this evening when we get access to it and the Rio.  However I tried using two PCs connected with a router, no rio available, and got network table server running on one machine receiving the Grip contours sent from the Grip PC1, even ran another viewer as a client on the Grip machine PC1 and received the contours back from the Viewer server on PC2 (now with two connections Grip and Viewer) using IPs of the machines, but was not able to see the published video on the second machine using the Grip PCs IP.

SamCarlberg, 2017-01-27:
@AustinShalit

WhyGitHub1, 2017-01-27:
Ok - I was using http://ip.  I tried just the ip address of the grip machine and it appears to be connecting.  Thanks for the comment about using IP address.

SamCarlberg, 2017-01-29:
@WhyGitHub1 If you've resolved the problem, please close this issue

WhyGitHub1, 2017-01-29:
Issue resolved thanks all

closed 2017-01-29 by WhyGitHub1