WPIRoboticsProjects/GRIP #805: GRIP Code Generation Capitalization Error

333fred, 2017-01-27:
_From @RaidingForPants on January 27, 2017 20:58_

When using the GRIP Computer Vision Pipeline to generate code, the function which overrides the virtual function "Process" in the parent class, VisionPipeline, is not capitalized, and is generated as "process".

In MyPipeline.h:
`public:
		MyPipeline();
		void process(cv::Mat source0);
		cv::Mat* gethsvThresholdOutput();
		cv::Mat* getcvDilateOutput();
`

In VisionPipeline.h:
`public:
  virtual ~VisionPipeline() = default;

  /**
   * Processes the image input and sets the result objects.
   * Implementations should make these objects accessible.
   */
  virtual void Process(cv::Mat& mat) = 0;
`

_Copied from original issue: wpilibsuite/allwpilib#473_

SamCarlberg, 2017-01-27:
This is a known issue and I've already started working on it. See #798

333fred, 2017-01-27:
Dupe of https://github.com/WPIRoboticsProjects/GRIP/issues/797.

closed 2017-01-27 by 333fred