WPIRoboticsProjects/GRIP #811: SimpleBlobDetector: Generated configuration in Java template differs from configuration used in GUI

mfkahn, 2017-01-29:
Our team could not figure out why a pipeline that appeared to generate a particular blob detection in the GRIP GUI did not generate the same results when used in the generated VisionPipeline java class. We used the exact same static input image and produced two completely different blob arrays.

I dug into the generated (Java) code and the OpenCV defaults, and here's the issue:

The generated code writes the following in the XML configuration:

```
			config.append("<filterByInertia>0</filterByInertia>\n");
			config.append("<filterByConvexity>0</filterByConvexity>\n");
```

In order to be identical to the SimpleBlobDetector instance used in the GRIP GUI (which uses the OpenCV parameter defaults), the template should be:

```
                        config.append("<filterByInertia>1</filterByInertia>\n");
			config.append("<minInertiaRatio>0.1</minInertiaRatio>\n");
			config.append("<maxInertiaRatio>" + Integer.MAX_VALUE + "</maxInertiaRatio>\n");
			config.append("<filterByConvexity>1</filterByConvexity>\n");
			config.append("<minConvexity>0.95</minConvexity>\n");
			config.append("<maxConvexity>" + Integer.MAX_VALUE + "</maxConvexity>\n");
```

closed 2017-01-30 by SamCarlberg

nightpool, 2017-01-30:
@SamCarlberg could this possibly also be the cause of #806?

SamCarlberg, 2017-01-30:
No, that's using contour detection. No blobs.

PR #812 (2017-01-30): Fix generated Java blob detection
Closes #811
commit 0bc10877 (2017-01-30): Fix generated Java blob detection (#812)

Closes #811