WPIRoboticsProjects/GRIP #815: If filtered contour area is 0, generated python divides by 0

brookshank, 2017-01-31:
Not a huge issue as setting the minimum area to 1 fixes. Could be part of a larger issue in #806 though?

SamCarlberg, 2017-01-31:
[Relevant line](https://github.com/WPIRoboticsProjects/GRIP/blob/master/ui/src/main/resources/edu/wpi/grip/ui/codegeneration/python/operations/Filter_Contours.vm#L35)

brookshank, 2017-01-31:
I wasn't sure if it was known or not, feel free to remove the issue.

SamCarlberg, 2017-01-31:
It wasn't. Thanks for letting us know

JLLeitschuh, 2017-02-01:
[image]

Thanks for letting us know.
Sam, what should we assume when the value is zero? That they want the whole image or that they want none of it?

SamCarlberg, 2017-02-01:
Assume 100% solidity. Our throw out zero-size contours as they aren't very useful anyway