WPIRoboticsProjects/GRIP #816: `:ui:run` works properly but `:ui:installDist` doesn't

rlee287, 2017-02-01:
The scripts created by `:ui:installDist` do not work on Windows.

`gradlew clean :ui:run` cleans before building and running as normal.

`gradlew clean :ui:installDist` cleans and claims successful build.

However, running the generated `ui.bat` creates the following error:

```
<snip>\GRIP\ui\build\install\ui\bin>ui.bat
Error: Could not find or load main class someRandomGiberish0.44752144612949274
```

```
> git describe --tags
v1.5.2

> git show HEAD
commit 8d3fa6cf705a8d37a055b99ebca21fbb9c746404
```

AustinShalit, 2017-02-01:
What are you trying to do?  The way to generate the native app bundle is this:
```
./gradlew jfxNative
```

rlee287, 2017-02-03:
I was trying to create a standalone `.jar`. Thanks for the advice. I found that `./gradlew jfxJar` works for this purpose.

closed 2017-02-03 by rlee287