WPIRoboticsProjects/GRIP #819: Opening file from command line on Ubuntu Linux fails with "GRIP Failed to launch JVM"

2239Technocrats, 2017-02-10:
We are an FRC team. We're using a Kangaroo + on which we have installed Ubuntu Linux using Rufus and a flashdrive.

We installed GRIP by getting the .deb package from online and then using
`sudo dpkg -i grip-1.5.2-x64.deb`

When I use the GRIP icon to open GRIP, or I use
`\opt\GRIP\GRIP` it works fine.

But, `\opt\GRIP\GRIP -f filename.grip` and `\opt\GRIP\GRIP --file filename.grip` both fail to launch, leaving me with the following messages:

```

Feb 09, 2017 6:14:09 PM edu.wpi.grip.ui.Main init
INFO: Launching GRIP in UI mode
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
platform: /Linux/amd64/
Feb 09, 2017 6:14:26 PM edu.wpi.grip.core.util.service.LoggingListener starting
INFO: [PipelineRunner] Starting
Feb 09, 2017 6:14:49 PM edu.wpi.grip.core.CoreCommandLineHelper loadFile
WARNING: Invalid file: SecondProject.grip
java.io.FileNotFoundException: SecondProject.grip (No such file or directory)
	at java.io.FileInputStream.open0(Native Method)
	at java.io.FileInputStream.open(FileInputStream.java:195)
	at java.io.FileInputStream.<init>(FileInputStream.java:138)
	at edu.wpi.grip.core.serialization.Project.open(Project.java:99)
	at edu.wpi.grip.core.CoreCommandLineHelper.loadFile(CoreCommandLineHelper.java:162)
	at edu.wpi.grip.ui.Main.start(Main.java:144)
	at com.sun.javafx.application.LauncherImpl.lambda$launchApplication1$162(LauncherImpl.java:863)
	at com.sun.javafx.application.PlatformImpl.lambda$runAndWait$175(PlatformImpl.java:326)
	at com.sun.javafx.application.PlatformImpl.lambda$null$173(PlatformImpl.java:295)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$174(PlatformImpl.java:294)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
	at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
	at com.sun.glass.ui.gtk.GtkApplication.lambda$null$49(GtkApplication.java:139)
	at java.lang.Thread.run(Thread.java:745)

Exception in Application start method
GRIP Error invoking method.
GRIP Failed to launch JVM

```

How can I open a file from the command line with GRIP in Ubuntu Linux on a Kangaroo Plus?

SamCarlberg, 2017-02-10:
You need to pass the full path, eg `/home/<user>/.../project.grip`

closed 2017-02-13 by SamCarlberg

JLLeitschuh, 2017-03-01:
Do we want to support using relative paths as well?