WPIRoboticsProjects/GRIP #824: Help with Running GRIP with OpenCV Strait from Raspberry Pi

Jetlogan, 2017-02-11:
I'm kind of a newbie with OpenCV and running Java programs. Does anyone know the command to run a GRIP Java file with OpenCV on a Raspberry Pi? I tried somethings but they didn't work. Thanks.

SamCarlberg, 2017-02-11:
https://github.com/WPIRoboticsProjects/GRIP-code-generation/tree/master/java

closed 2017-02-13 by SamCarlberg

Jetlogan, 2017-02-14:
Sorry for the long response. When I do the command:
`java -Djava.library.path=${OPENCV_LIB} -jar GripPipeline.java`
I get:
`Error: Invalid or corrupt jarfile GripPipeline.java`

The GripPipeline.jar is in the builds folder of OpenCV. I have tried other .jar files from the GitHub link you provided. Here are my variables in my bashrc. Do you know why it's doing this?

SamCarlberg, 2017-02-14:
1. You have to compile the java file to a class to be able to run it
2. GRIP doesn't generate a runnable file. You have to include it in its own program.

Jetlogan, 2017-02-14:
Others have told me that you gotta use OpenCV, and I thought you told me the the GRIP.jar was depreciated.

SamCarlberg, 2017-02-15:
The _deploy_ was deprecated. Code generation is entirely different. Basically, it's intended to generate a java file into an existing project that you handle like any other java program. You're trying to run just the java file it's generating without compiling it or using it in a program.

Jetlogan, 2017-02-15:
So how should we put the .jar file GRIP generates in a executable Java file.

SamCarlberg, 2017-02-15:
1. Create a new Java project in the IDE of your choice (Eclipse, Netbeans, ...)
2. Create a main class. It can be empty, you'll fill it in later
3. Generate Java code from GRIP. Hit the browse button and choose to the directory where the main file is. Make sure to set the package name, too.
4. Hit "Okay" to generate the Java file
5. Go back to the main file and add the main method. You can use the functions of the generated class to get the outputs of the steps you want.