WPIRoboticsProjects/GRIP #826: Symantec flagging virus

JohnDuquette, 2017-02-15:
Symantec is flagging a flagging GRIP-v1.5.2-x86_32.exe as a WS.Reputation.1 risk when it executes.  It gets quarantined 'because not enough Symantec gurus have scanned it'.  Here are the details:  

http://securityresponse.symantec.com/security_response/writeup.jsp?docid=2010-051308-1854-99&vid=4294919973

GRIP-v1.5.2-x64.exe has no issues with Symantec.  

I thought you'd want to know.  Cool tool!

SamCarlberg, 2017-02-15:
Interesting that it flags the 32-bit version (which @AustinShalit builds manually) but not the 64-bit one that our CI tools build for us.

AustinShalit, 2017-02-15:
*innocent face*

This could be that less people use the 32-bit build?  Could also be an artifact of building in my VM. This will probably be solved as soon as we figure out how to get AppVeyor to build 32-bit

JLLeitschuh, 2017-02-17:
This makes sense, we aren't a heavily used application so I'm not shocked that we don't have any sort of reputation with Symantec.

I think this could be flagged as a "wontfix", even though its more of a "can't fix"