WPIRoboticsProjects/GRIP #827: GRIP will not import into newest version of Eclipse

KevinNJ, 2017-02-15:
Hello,

I downloaded and installed Eclipse Neon.2 version 4.6.2 and attempted to follow the instructions for building GRIP from source in Eclipse from here (https://github.com/WPIRoboticsProjects/GRIP/wiki/Building-GRIP-from-Source-in-Eclipse)

Eclipse 4.6.2 comes preloaded with Gradle support, so I skipped to the Importing GRIP Gradle Project step.  When I try to finish the import, I get the following errors:

Synchronize Gradle builds with workspace failed due to an error connecting to the Gradle build.
Could not fetch model of type 'EclipseProject' using Gradle distribution 'https://services.gradle.org/distributions/gradle-3.2.1-all.zip'.

Could not get unknown property 'generated' for SourceSet container.
org.gradle.tooling.GradleConnectionException: Could not fetch model of type 'EclipseProject' using Gradle distribution 'https://services.gradle.org/distributions/gradle-3.2.1-all.zip'.
	at org.gradle.tooling.internal.consumer.ExceptionTransformer.transform(ExceptionTransformer.java:55)
	at org.gradle.tooling.internal.consumer.ExceptionTransformer.transform(ExceptionTransformer.java:29)
	at org.gradle.tooling.internal.consumer.ResultHandlerAdapter.onFailure(ResultHandlerAdapter.java:41)
	at org.gradle.tooling.internal.consumer.async.DefaultAsyncConsumerActionExecutor$1$1.run(DefaultAsyncConsumerActionExecutor.java:57)
	at org.gradle.internal.concurrent.ExecutorPolicy$CatchAndRecordFailures.onExecute(ExecutorPolicy.java:54)
	at org.gradle.internal.concurrent.StoppableExecutorImpl$1.run(StoppableExecutorImpl.java:40)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)
	at org.gradle.tooling.internal.consumer.BlockingResultHandler.getResult(BlockingResultHandler.java:46)
	at org.gradle.tooling.internal.consumer.DefaultModelBuilder.get(DefaultModelBuilder.java:51)
	at com.gradleware.tooling.toolingclient.internal.DefaultToolingClient.executeAndWait(DefaultToolingClient.java:106)
	at com.gradleware.tooling.toolingclient.internal.DefaultModelRequest.executeAndWait(DefaultModelRequest.java:79)
	at com.gradleware.tooling.toolingmodel.repository.internal.BaseModelRepository$1.get(BaseModelRepository.java:95)
	at com.gradleware.tooling.toolingmodel.repository.internal.BaseModelRepository.executeAndWait(BaseModelRepository.java:163)
	at com.gradleware.tooling.toolingmodel.repository.internal.BaseModelRepository.access$000(BaseModelRepository.java:41)
	at com.gradleware.tooling.toolingmodel.repository.internal.BaseModelRepository$2.call(BaseModelRepository.java:121)
	at com.google.common.cache.LocalCache$LocalManualCache$1.load(LocalCache.java:4724)
	at com.google.common.cache.LocalCache$LoadingValueReference.loadFuture(LocalCache.java:3522)
	at com.google.common.cache.LocalCache$Segment.loadSync(LocalCache.java:2315)
	at com.google.common.cache.LocalCache$Segment.lockedGetOrLoad(LocalCache.java:2278)
	at com.google.common.cache.LocalCache$Segment.get(LocalCache.java:2193)
	at com.google.common.cache.LocalCache.get(LocalCache.java:3932)
	at com.google.common.cache.LocalCache$LocalManualCache.get(LocalCache.java:4721)
	at com.gradleware.tooling.toolingmodel.repository.internal.BaseModelRepository.getFromCache(BaseModelRepository.java:138)
	at com.gradleware.tooling.toolingmodel.repository.internal.BaseModelRepository.executeRequest(BaseModelRepository.java:117)
	at com.gradleware.tooling.toolingmodel.repository.internal.BaseModelRepository.executeRequest(BaseModelRepository.java:88)
	at com.gradleware.tooling.toolingmodel.repository.internal.DefaultSingleBuildModelRepository.fetchEclipseGradleBuild(DefaultSingleBuildModelRepository.java:185)
	at org.eclipse.buildship.core.workspace.internal.DefaultModelProvider.fetchEclipseGradleBuild(DefaultModelProvider.java:53)
	at org.eclipse.buildship.core.workspace.internal.SynchronizeGradleBuildsJob.synchronizeBuild(SynchronizeGradleBuildsJob.java:77)
	at org.eclipse.buildship.core.workspace.internal.SynchronizeGradleBuildsJob.runToolingApiJob(SynchronizeGradleBuildsJob.java:69)
	at org.eclipse.buildship.core.util.progress.ToolingApiJob$1.run(ToolingApiJob.java:73)
	at org.eclipse.buildship.core.util.progress.ToolingApiInvoker.invoke(ToolingApiInvoker.java:63)
	at org.eclipse.buildship.core.util.progress.ToolingApiJob.run(ToolingApiJob.java:70)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:55)
Caused by: groovy.lang.MissingPropertyException: Could not get unknown property 'generated' for SourceSet container.
	at org.gradle.internal.metaobject.AbstractDynamicObject.getMissingProperty(AbstractDynamicObject.java:92)
	at org.gradle.internal.metaobject.AbstractDynamicObject.getProperty(AbstractDynamicObject.java:62)
	at org.gradle.api.internal.tasks.DefaultSourceSetContainer_Decorated.getProperty(Unknown Source)
	at org.codehaus.groovy.runtime.callsite.PogoGetPropertySite.getProperty(PogoGetPropertySite.java:50)
	at org.codehaus.groovy.runtime.callsite.AbstractCallSite.callGetProperty(AbstractCallSite.java:296)
	at build_6ov8zwme2a0kzr0bpsiw7dw2k$_run_closure7$_closure46$_closure56.doCall(C:\Users\kevin.nelson\Downloads\GRIP-master\GRIP-master\build.gradle:309)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.codehaus.groovy.reflection.CachedMethod.invoke(CachedMethod.java:93)
	at groovy.lang.MetaMethod.doMethodInvoke(MetaMethod.java:325)
	at org.codehaus.groovy.runtime.metaclass.ClosureMetaClass.invokeMethod(ClosureMetaClass.java:294)
	at groovy.lang.MetaClassImpl.invokeMethod(MetaClassImpl.java:1024)
	at groovy.lang.Closure.call(Closure.java:414)
	at groovy.lang.Closure.call(Closure.java:430)
	at org.gradle.api.internal.ClosureBackedAction.execute(ClosureBackedAction.java:70)
	at org.gradle.util.ConfigureUtil.configureTarget(ConfigureUtil.java:160)
	at org.gradle.util.ConfigureUtil.configure(ConfigureUtil.java:106)
	at org.gradle.util.ConfigureUtil$1.execute(ConfigureUtil.java:123)
	at org.gradle.listener.ActionBroadcast.execute(ActionBroadcast.java:39)
	at org.gradle.plugins.ide.eclipse.model.EclipseClasspath.mergeXmlClasspath(EclipseClasspath.java:313)
	at org.gradle.plugins.ide.internal.tooling.EclipseModelBuilder.populate(EclipseModelBuilder.java:153)
[... 41 lines cut ...]
	at org.gradle.launcher.daemon.server.api.DaemonCommandExecution.proceed(DaemonCommandExecution.java:120)
	at org.gradle.launcher.daemon.server.exec.ResetDeprecationLogger.execute(ResetDeprecationLogger.java:26)
	at org.gradle.launcher.daemon.server.api.DaemonCommandExecution.proceed(DaemonCommandExecution.java:120)
	at org.gradle.launcher.daemon.server.exec.RequestStopIfSingleUsedDaemon.execute(RequestStopIfSingleUsedDaemon.java:34)
	at org.gradle.launcher.daemon.server.api.DaemonCommandExecution.proceed(DaemonCommandExecution.java:120)
	at org.gradle.launcher.daemon.server.exec.ForwardClientInput$2.call(ForwardClientInput.java:74)
	at org.gradle.launcher.daemon.server.exec.ForwardClientInput$2.call(ForwardClientInput.java:72)
	at org.gradle.util.Swapper.swap(Swapper.java:38)
	at org.gradle.launcher.daemon.server.exec.ForwardClientInput.execute(ForwardClientInput.java:72)
	at org.gradle.launcher.daemon.server.api.DaemonCommandExecution.proceed(DaemonCommandExecution.java:120)
	at org.gradle.launcher.daemon.server.exec.LogAndCheckHealth.execute(LogAndCheckHealth.java:55)
	at org.gradle.launcher.daemon.server.api.DaemonCommandExecution.proceed(DaemonCommandExecution.java:120)
	at org.gradle.launcher.daemon.server.exec.LogToClient.doBuild(LogToClient.java:60)
	at org.gradle.launcher.daemon.server.exec.BuildCommandOnly.execute(BuildCommandOnly.java:36)
	at org.gradle.launcher.daemon.server.api.DaemonCommandExecution.proceed(DaemonCommandExecution.java:120)
	at org.gradle.launcher.daemon.server.exec.EstablishBuildEnvironment.doBuild(EstablishBuildEnvironment.java:72)
	at org.gradle.launcher.daemon.server.exec.BuildCommandOnly.execute(BuildCommandOnly.java:36)
	at org.gradle.launcher.daemon.server.api.DaemonCommandExecution.proceed(DaemonCommandExecution.java:120)
	at org.gradle.launcher.daemon.server.exec.StartBuildOrRespondWithBusy$1.run(StartBuildOrRespondWithBusy.java:50)
	at org.gradle.launcher.daemon.server.DaemonStateCoordinator$1.run(DaemonStateCoordinator.java:293)
	at org.gradle.internal.concurrent.ExecutorPolicy$CatchAndRecordFailures.onExecute(ExecutorPolicy.java:54)
	at org.gradle.internal.concurrent.StoppableExecutorImpl$1.run(StoppableExecutorImpl.java:40)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)

KevinNJ, 2017-02-15:
P.S.
I was able to get the project working in IntelliJ using the instructions found here:
https://github.com/WPIRoboticsProjects/GRIP/wiki/Setting-up-build-tools

Just thought you should know that some instructions on the WIKI don't work.

Thanks for the great tool!

rlee287, 2017-03-05:
I was the one who originally set up the wiki instructions. When trying to open the GRIP project in Eclipse a few weeks ago, I found that Gradle had broken. I ran into multiple issues trying to use Eclipse gradle integration for other projects, so I ended up having to use the Gradle command line. I believe that the Eclipse Gradle integration broke sometime after I set up that wiki page.

I will add a note to the wiki article noting that the instructions no longer work.