WPIRoboticsProjects/GRIP #829: Error When Running a .jar Grip File

Jetlogan, 2017-02-18:
`pi@raspberrypi:~/opencv-3.1.0 $ java -Djava.library.path=${OPENCV_LIB} -jar build/MyCoolGripJar.jar
Exception in thread "main" java.lang.UnsupportedClassVersionError: something/main : Unsupported major.minor version 5                                                                                                                        2.0
        at java.lang.ClassLoader.defineClass1(Native Method)
        at java.lang.ClassLoader.defineClass(ClassLoader.java:800)
        at java.security.SecureClassLoader.defineClass(SecureClassLoader.java:142)
        at java.net.URLClassLoader.defineClass(URLClassLoader.java:449)
        at java.net.URLClassLoader.access$100(URLClassLoader.java:71)
        at java.net.URLClassLoader$1.run(URLClassLoader.java:361)
        at java.net.URLClassLoader$1.run(URLClassLoader.java:355)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.net.URLClassLoader.findClass(URLClassLoader.java:354)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:425)
        at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:308)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:358)
        at sun.launcher.LauncherHelper.checkAndLoadMain(LauncherHelper.java:482)
`

I'm not the best at Java. Anyone know whats going on? I'm using OpenCV 3.1.0 on both sides.

SamCarlberg, 2017-02-18:
What version of Java did you use to build the jar? What version are you running the jar with?

Jetlogan, 2017-02-18:
I'm using Java Development Kit 8u11.
On the Raspberry Pi JDK 7.

JLLeitschuh, 2017-02-22:
`Unsupported major.minor version 5`
That's not a valid version of java. Usually you'll see an error like `Unsupported major.minor version 52.0`.

Are you sure that `build/MyCoolGripJar.jar` is a valid JAR?
Also, are you trying to run GRIP or code generated with GRIP?