WPIRoboticsProjects/GRIP #836: ROS publishing operations don't work

SamCarlberg, 2017-03-15:
Trying to use any ROS operation will fail. Looks like the internal publisher never gets set

@JLLeitschuh

JLLeitschuh, 2017-03-16:
I'm not sure if it actually works on a system that doesn't have ROS installed.

tomatac, 2017-12-14:
It does not work in ROS Kinetic:
java.util.NoSuchElementException: No value present
	at java.util.Optional.get(Optional.java:135)
	at edu.wpi.grip.core.operations.network.ros.ROSManager$ROSNetworkPublisher.publish(ROSManager.java:127)
	at edu.wpi.grip.core.operations.network.ros.ROSManager$ROSNetworkPublisher.publish(ROSManager.java:89)
	at edu.wpi.grip.core.operations.network.ros.ROSPublishOperation.doPublish(ROSPublishOperation.java:58)
	at edu.wpi.grip.core.operations.network.NetworkPublishOperation.perform(NetworkPublishOperation.java:75)
	at edu.wpi.grip.core.metrics.Timer.time(Timer.java:102)
	at edu.wpi.grip.core.Step.runPerform(Step.java:138)
	at edu.wpi.grip.core.PipelineRunner.runPipeline(PipelineRunner.java:212)
	at edu.wpi.grip.core.PipelineRunner.access$100(PipelineRunner.java:43)
	at edu.wpi.grip.core.PipelineRunner$1.lambda$runOneIteration$1(PipelineRunner.java:91)
	at edu.wpi.grip.core.metrics.Timer.time(Timer.java:102)
	at edu.wpi.grip.core.PipelineRunner$1.runOneIteration(PipelineRunner.java:91)
	at com.google.common.util.concurrent.AbstractScheduledService$ServiceDelegate$Task.run(AbstractScheduledService.java:188)
	at com.google.common.util.concurrent.Callables$4.run(Callables.java:122)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:308)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$301(ScheduledThreadPoolExecutor.java:180)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:294)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)

tomatac, 2017-12-14:
Is there a workaround to communicate with ROS?

JLLeitschuh, 2017-12-15:
You are probably the first person to actually try to use the ROS stuff inside of GRIP since I've written it.
(Pretty cool in my book).
Do you think that the version of ROS Java that we are using is out of date?

I'm happy to work with you to get this fixed.