WPIRoboticsProjects/GRIP #846: Error setting up build tools

ajbt200128, 2017-04-14:
When setting up build tools there are 2 problems I run into. One is that maven says there are unindexed remote maven repositiries found, and when I try to update them(http://first.wpi.edu/FRC/roborio/maven/development and http://first.wpi.edu/FRC/roborio/maven/release) it gives an error saying "nexus-maven-repository-index.properties does not exist". The second is when I try to build it gives errors saying packages "edu.wpi.grip.generated.opencv_core.enumeration.CmpTypesEnum" and others of the same type do not exist

JLLeitschuh, 2017-04-14:
Is this in intellij?
The unindexed remote maven repositories problem is not really a problem just a warning. It can be safely ignored.

ajbt200128, 2017-04-15:
The error still persists even after running `./gradlew generateCodeFromSource`. This is in intellij, also in the gradle projects window there are errors which say "unable to resolve ..." for NetworkTables, OpenCV, rosjava, and grip_msgs. Could me being on windows have anything to do with it?

JLLeitschuh, 2017-04-18:
@SamCarlberg Have you seen this before??

SamCarlberg, 2017-04-18:
I haven't seen this before. Just running `./gradlew :ui:run` (or `gradlew.bat :ui:run` in this case) from a fresh clone should work as long as you have an internet connection

ajbt200128, 2017-04-24:
I can build the gradle project but the problem is if I try to edit any code it will tell me that there are unresolved symbols with the openv library imports, and in the gradle window under :ui/Source Sets/main/Dependencies it says those libraries cannot be resolved

closed 2023-10-26 by ajbt200128