WPIRoboticsProjects/GRIP #856: Create GRIP Plugin for WPILib

AustinShalit, 2017-07-08:
We should create a WPILib plugin for grip that will allow teams to decode GRIP NetworkTable data as a single object.  That way all of the data values are synchronized with each other.

JLLeitschuh, 2017-07-10:
Can you provide a few more details on what you are thinking? How would this be used? What changes would be required from within the GRIP code base?

AustinShalit, 2017-07-10:
GRIP would put data in NetworkTables as a Raw type.  The plugin would decode the information and provide it to teams.  That way there are no synchronization issues with multiple NetworkTable arrays

JLLeitschuh, 2017-07-10:
Would this be a new step type? Would this replace the existing steps?

AustinShalit, 2017-07-10:
This would be an additional checkbox on the NTPublish_XReport steps.

JLLeitschuh, 2017-07-11:
I'd like a bit more of a fleshed out description before a PR is attempted. What is the high level goal here?
Would this be an object that ships with WPILib?
How would you have the synchronization guarantees? How would you support forwards/backwards compatibility?
Would the object be JSON or a byte array that gets serialized/deserialized?