WPIRoboticsProjects/GRIP #861: Errors upon adding webcam

StefanCamilleri, 2017-08-17:
Hardware: VF0680 Live! Cam Socalize HD 1080
Using the GUI in Win7 64bit Pro.
1. Start GRIP
2. Add Source -> Webcam (0, 1, 2, 3...)

#From the Log File:
ug 17, 2017 4:42:54 PM edu.wpi.grip.core.util.service.LoggingListener starting
INFO: [PipelineRunner] Starting
Aug 17, 2017 4:42:59 PM edu.wpi.grip.core.util.service.LoggingListener starting
INFO: [CameraSource] Starting
Aug 17, 2017 4:42:59 PM edu.wpi.grip.core.events.EventLogger eventPosted
INFO: Event on thread 'JavaFX Application Thread': SourceAddedEvent{source=CameraSource{name=Webcam 0}}
Aug 17, 2017 4:43:00 PM edu.wpi.grip.core.util.service.LoggingListener starting
INFO: [CameraSource] Starting
Aug 17, 2017 4:43:00 PM edu.wpi.grip.core.util.service.LoggingListener failed
SEVERE: [CameraSource] Failed from: RUNNING
edu.wpi.grip.core.sources.GrabberService$GrabberServiceException: [Webcam 0] Failed to grab image
	at edu.wpi.grip.core.sources.GrabberService.runOneGrab(GrabberService.java:75)
	at edu.wpi.grip.core.sources.GrabberService.run(GrabberService.java:64)
	at com.google.common.util.concurrent.AbstractExecutionThreadService$1$2.run(AbstractExecutionThreadService.java:66)
	at com.google.common.util.concurrent.Callables$4.run(Callables.java:122)
	at java.lang.Thread.run(Thread.java:745)
Caused by: org.bytedeco.javacv.FrameGrabber$Exception: videoInput.getPixels() Error: Could not get pixels.
	at org.bytedeco.javacv.VideoInputFrameGrabber.grab(VideoInputFrameGrabber.java:171)
	at edu.wpi.grip.core.sources.GrabberService.runOneGrab(GrabberService.java:73)
	... 4 more

#From the dialog pop-up (which only occurred once)
Thread[JavaFX Application Thread,5,main] threw an exception
## Stack Trace:
```java
java.lang.IllegalStateException: Can not start while in state STARTING. Must be in one of states: NEW, FAILED, TERMINATED
	at edu.wpi.grip.core.util.service.AutoRestartingService.startAsync(AutoRestartingService.java:123)
	at edu.wpi.grip.core.util.service.AutoRestartingService.startAsync(AutoRestartingService.java:108)
	at edu.wpi.grip.core.sources.CameraSource.startAsync(CameraSource.java:270)
	at edu.wpi.grip.core.sources.CameraSource.startAsync(CameraSource.java:46)
	at edu.wpi.grip.ui.components.StartStoppableButton.lambda$new$1(StartStoppableButton.java:64)
	at com.sun.javafx.event.CompositeEventHandler$NormalEventFilterRecord.handleCapturingEvent(CompositeEventHandler.java:282)
	at com.sun.javafx.event.CompositeEventHandler.dispatchCapturingEvent(CompositeEventHandler.java:98)
	at com.sun.javafx.event.EventHandlerManager.dispatchCapturingEvent(EventHandlerManager.java:223)
	at com.sun.javafx.event.EventHandlerManager.dispatchCapturingEvent(EventHandlerManager.java:180)
	at com.sun.javafx.event.CompositeEventDispatcher.dispatchCapturingEvent(CompositeEventDispatcher.java:43)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:52)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.EventUtil.fireEventImpl(EventUtil.java:74)
	at com.sun.javafx.event.EventUtil.fireEvent(EventUtil.java:54)
	at javafx.event.Event.fireEvent(Event.java:198)
	at javafx.scene.Scene$MouseHandler.process(Scene.java:3757)
	at javafx.scene.Scene$MouseHandler.access$1500(Scene.java:3485)
	at javafx.scene.Scene.impl_processMouseEvent(Scene.java:1762)
	at javafx.scene.Scene$ScenePeerListener.mouseEvent(Scene.java:2494)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(GlassViewEventHandler.java:380)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(GlassViewEventHandler.java:294)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler.lambda$handleMouseEvent$354(GlassViewEventHandler.java:416)
	at com.sun.javafx.tk.quantum.QuantumToolkit.runWithoutRenderLock(QuantumToolkit.java:389)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler.handleMouseEvent(GlassViewEventHandler.java:415)
	at com.sun.glass.ui.View.handleMouseEvent(View.java:555)
	at com.sun.glass.ui.View.notifyMouse(View.java:937)
	at com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
	at com.sun.glass.ui.win.WinApplication.lambda$null$148(WinApplication.java:191)
	at java.lang.Thread.run(Thread.java:745)

```
## System Info:

Property Name | Property
 ----- | -----
GRIP Version | v1.5.2
java.version | 1.8.0_101
javafx.version | 8.0.101
os.name | Windows 7
os.version | 6.1
os.arch | amd64

JLLeitschuh, 2017-08-17:
Do you get this problem a lot or was this a one time thing?

StefanCamilleri, 2017-08-17:
I have never not had this problem. I just installed GRIP today.

JLLeitschuh, 2017-08-17:
Ah, can you get the webcam to be picked up by any other programs? Does it work for example with hangouts?

StefanCamilleri, 2017-08-17:
It actually works when I use it with the OpenCV example programs in eclipse found here:
https://opencv-java-tutorials.readthedocs.io/en/latest/

However, it did not work with SimpleCV, and my understanding was that only certain hardware worked with SimpleCV.

Just now I tried testing my webcam with Skype and it said that it was in use by another program. GRIP had been sitting frozen in the background. Shutting down GRIP immediately allowed it to produce an image in Skype.

JLLeitschuh, 2017-08-18:
So on windows we don't use the OpenCV frame grabber:
https://github.com/WPIRoboticsProjects/GRIP/blob/master/core/src/main/java/edu/wpi/grip/core/sources/CameraSource.java#L404-L405

Instead we use the [VideoInputFrameGrabber](https://github.com/bytedeco/javacv/blob/master/src/main/java/org/bytedeco/javacv/VideoInputFrameGrabber.java).

We don't really expose many of the knobs that the API exposes and perhaps this is why this isn't working.

One possible solution to get this working with your camera is to run the camera through [cscore](https://github.com/wpilibsuite/cscore) which will create an http server that's hosting an MJPEG stream with your camera as a source.

Another option you could try is checking out the GRIP source code and commenting out the lines I liked to above so that GRIP tries to load your camera with the `OpenCVFrameGrabber`.

If my second solution works for you then that may mean we want to expose the choice of frame grabber in the UI.

StefanCamilleri, 2017-08-18:
So gradlew.bat :ui:run compiles fine with no changes.

When I make this edit,

if (StandardSystemProperty.OS_NAME.value().contains("Windows")) {
        //return new VideoInputFrameGrabber(deviceNumber);
	return new OpenCVFrameGrabber(deviceNumber);
      } else {
        return new OpenCVFrameGrabber(deviceNumber);
      }

the build fails:

C:\Users\Stefan\source\repos\GRIP>gradlew.bat :ui:run

> Configure project :
Gradle now uses separate output directories for each JVM language, but this buil
d assumes a single directory for all classes from a source set. This behaviour h
as been deprecated and is scheduled to be removed in Gradle 5.0
        at build_1tc2hgaxbusqoc3gapot3d6me$_run_closure10$_closure72.doCall(C:\U
sers\Stefan\source\repos\GRIP\build.gradle:396)

> Task :core:compileJava
C:\Users\Stefan\source\repos\GRIP\core\src\main\java\edu\wpi\grip\core\sources\V
ideoFileSource.java:133: warning: [FutureReturnValueIgnored] Return value of met
hods returning Future must be checked. Ignoring returned Futures suppresses exce
ptions thrown from the code that completes the Future.
        manualGrabberService.submit(this::grabNextFrame);
                                   ^
    (see http://errorprone.info/bugpattern/FutureReturnValueIgnored)
  Did you mean to remove this line?
C:\Users\Stefan\source\repos\GRIP\core\src\main\java\edu\wpi\grip\core\sources\V
ideoFileSource.java:223: warning: [FutureReturnValueIgnored] Return value of met
hods returning Future must be checked. Ignoring returned Futures suppresses exce
ptions thrown from the code that completes the Future.
    manualGrabberService.submit(this::grabNextFrame);
                               ^
    (see http://errorprone.info/bugpattern/FutureReturnValueIgnored)
  Did you mean to remove this line?
C:\Users\Stefan\source\repos\GRIP\core\src\generated\java\edu\wpi\grip\generated
\opencv_core\enumeration\Enum.java:5: warning: [JavaLangClash] edu.wpi.grip.gene
rated.opencv_core.enumeration.Enum clashes with java.lang.Enum
public enum Enum {
       ^

    (see http://errorprone.info/bugpattern/JavaLangClash)
C:\Users\Stefan\source\repos\GRIP\core\build\classes\java\main\edu\wpi\grip\core
\operations\composite\AutoValue_ContoursReport_Contour.java:94: warning: [Narrow
ingCompoundAssignment] Compound assignments from long to int hide lossy casts
    h ^= (Double.doubleToLongBits(this.area) >>> 32) ^ Double.doubleToLongBits(t
his.area);
      ^
    (see http://errorprone.info/bugpattern/NarrowingCompoundAssignment)
  Did you mean 'h = (int) (h ^ ((Double.doubleToLongBits(this.area) >>> 32) ^ Do
uble.doubleToLongBits(this.area)));'?
C:\Users\Stefan\source\repos\GRIP\core\build\classes\java\main\edu\wpi\grip\core
\operations\composite\AutoValue_ContoursReport_Contour.java:96: warning: [Narrow
ingCompoundAssignment] Compound assignments from long to int hide lossy casts
    h ^= (Double.doubleToLongBits(this.centerX) >>> 32) ^ Double.doubleToLongBit
s(this.centerX);
      ^
    (see http://errorprone.info/bugpattern/NarrowingCompoundAssignment)
  Did you mean 'h = (int) (h ^ ((Double.doubleToLongBits(this.centerX) >>> 32) ^
 Double.doubleToLongBits(this.centerX)));'?
C:\Users\Stefan\source\repos\GRIP\core\build\classes\java\main\edu\wpi\grip\core
\operations\composite\AutoValue_ContoursReport_Contour.java:98: warning: [Narrow
ingCompoundAssignment] Compound assignments from long to int hide lossy casts
    h ^= (Double.doubleToLongBits(this.centerY) >>> 32) ^ Double.doubleToLongBit
s(this.centerY);
      ^
[... 34 lines cut ...]
9 warnings

> Task :ui:compileJava
error: Bad service configuration file, or exception thrown while constructing Pr
ocessor object: javax.annotation.processing.Processor: Error reading configurati
on file
1 error

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':ui:compileJava'.
> Compilation failed with exit code 1; see the compiler error output for details
.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug
option to get more log output.

BUILD FAILED in 36s
7 actionable tasks: 4 executed, 3 up-to-date

C:\Users\Stefan\source\repos\GRIP>

Do I need to turn off some unit tests or something? I'm not sure how.

JLLeitschuh, 2017-08-18:
This seems to be a bug in our build. I have no idea why this happening.
Try running a `gradle.bat clean` before running `gradle.bat :ui:run`.

StefanCamilleri, 2017-08-18:
Ok, doing a clean allowed it to compile. I am able to add the camera and no exceptions are thrown, but this is what I see. Those little artifacts flicker (at about 5hz) in the same spot.
[image]

JLLeitschuh, 2017-08-19:
So your camera isn't showing anything (besides the flickering) or is complete blackness what you'd expect it to show?

@PeterJohnson Since you have experience with cameras do you have any thoughts on this?

StefanCamilleri, 2017-08-19:
I have never used this program before--I am assuming that I am supposed to get a live picture.

gazialankus, 2018-02-01:
I'm getting the exact build error as https://github.com/WPIRoboticsProjects/GRIP/issues/861#issuecomment-323435618 above when I edit the summary string in the `@Description` of `FindBlobsOperation`

Here's the only workaround I've found so far: 

1. Close IntelliJ (otherwise it won't work)
2. Run ./gradlew clean
3. Reopen IntelliJ

It would be nice if there would be a better solution to this.