WPIRoboticsProjects/GRIP #862: Crash while grabbing from cscore

AustinShalit, 2017-08-18:
```
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x00007ff9d549441e, pid=15624, tid=0x0000000000002ec0
#
# JRE version: Java(TM) SE Runtime Environment (8.0_92-b14) (build 1.8.0_92-b14)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (25.92-b14 mixed mode windows-amd64 compressed oops)
# Problematic frame:
# C  [opencv_imgcodecs300.dll+0x4441e]
#
# Failed to write core dump. Minidumps are not enabled by default on client versions of Windows
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

---------------  T H R E A D  ---------------

Current thread (0x0000000025bbf800):  JavaThread "IP Camera 192.168.1.121 Service" [_thread_in_native, id=11968, stack(0x000000002e430000,0x000000002e530000)]

siginfo: ExceptionCode=0xc0000005, reading address 0x000000002ebf639f

Registers:
RAX=0x0000000000000002, RBX=0x0000000000017501, RCX=0x000000002e52e950, RDX=0x000000002a737897
RSP=0x000000002e52e8e0, RBP=0x000000002a737897, RSI=0x000000002f6f1100, RDI=0x000000002ebf639f
R8 =0x0000000000000002, R9 =0x0000000000000000, R10=0x000000002a737897, R11=0x00007ff9d5450000
R12=0x0000000000000000, R13=0x000000002f24b5d0, R14=0x0000000000000002, R15=0x000000002e52e950
RIP=0x00007ff9d549441e, EFLAGS=0x0000000000010202

Top of Stack: (sp=0x000000002e52e8e0)
0x000000002e52e8e0:   00000000ffffffc6 0000000000000000
0x000000002e52e8f0:   0000000000000000 0000000000000000
0x000000002e52e900:   000000002f2c0aa8 0000000000000040
0x000000002e52e910:   000000002f24b6c8 00007ff9d549304c
0x000000002e52e920:   0000000000000003 000000002e52e999
0x000000002e52e930:   000000002f24c150 0000000000000006
0x000000002e52e940:   0000000000000009 0000000000000000
0x000000002e52e950:   000000002ebf639f 0000000000017501
0x000000002e52e960:   0000001c2a737897 000000002f6f1100
0x000000002e52e970:   0000000000000001 000000002f6f12d8
0x000000002e52e980:   000000002f6f1100 ffffffffffb5a43c
0x000000002e52e990:   00000000000000b0 ffffff8e2f24b5f0
0x000000002e52e9a0:   fffffffe00000000 0000000000000001
0x000000002e52e9b0:   0000bdcbae25b3e4 0000000000000000
0x000000002e52e9c0:   000000000000004f 000000002f6f1100
0x000000002e52e9d0:   000000002f24b828 0000000000000000 

Instructions: (pc=0x00007ff9d549441e)
0x00007ff9d54943fe:   00 00 48 85 db 75 19 48 8b 46 28 48 8b ce ff 50
0x00007ff9d549440e:   18 84 c0 74 65 48 8b 5e 28 48 8b 3b 48 8b 5b 08
0x00007ff9d549441e:   0f b6 07 48 ff cb 48 ff c7 3d ff 00 00 00 75 39
0x00007ff9d549442e:   66 90 48 85 db 75 19 48 8b 46 28 48 8b ce ff 50 

Register to memory mapping:

RAX=0x0000000000000002 is an unknown value
RBX=0x0000000000017501 is an unknown value
RCX=0x000000002e52e950 is pointing into the stack for thread: 0x0000000025bbf800
RDX=0x000000002a737897 is an unknown value
RSP=0x000000002e52e8e0 is pointing into the stack for thread: 0x0000000025bbf800
RBP=0x000000002a737897 is an unknown value
RSI=0x000000002f6f1100 is an unknown value
RDI=0x000000002ebf639f is an unknown value
R8 =0x0000000000000002 is an unknown value
R9 =0x0000000000000000 is an unknown value
R10=0x000000002a737897 is an unknown value
R11=0x00007ff9d5450000 is an unknown value
[... 358 lines cut ...]
java_class_path (initial): C:\Users\VisitorDL1\Downloads\GRIP-master\GRIP-master\ui\build\classes\java\main;C:\Users\VisitorDL1\Downloads\GRIP-master\GRIP-master\ui\build\resources\main;C:\Users\VisitorDL1\Downloads\GRIP-master\GRIP-master\ui\preloader\build\libs\preloader-v0.0.0.jar;C:\Users\VisitorDL1\.gradle\caches\modules-2\files-2.1\com.google.code.findbugs\annotations\3.0.1\fc019a2216218990d64dfe756e7aa20f0069dea2\annotations-3.0.1.jar;C:\Users\VisitorDL1\Downloads\GRIP-master\GRIP-master\core\build\libs\core-v0.0.0-all.jar;C:\Users\VisitorDL1\.gradle\caches\modules-2\files-2.1\org.controlsfx\controlsfx\8.40.11\61694ec2668cc69116e970c7513aa3501179bc00\controlsfx-8.40.11.jar;C:\Users\VisitorDL1\.gradle\caches\modules-2\files-2.1\com.hierynomus\sshj\0.16.0\2c22fecbf6a7fb09ab8addc627b8fa2d8526d879\sshj-0.16.0.jar;C:\Users\VisitorDL1\.gradle\caches\modules-2\files-2.1\org.apache.velocity\velocity\1.7\2ceb567b8f3f21118ecdec129fe1271dbc09aa7a\velocity-1.7.jar;C:\Users\VisitorDL1\.gradle\caches\modules-2\files-2.1\net.jcip\jcip-annotations\1.0\afba4942caaeaf46aab0b976afd57cc7c181467e\jcip-annotations-1.0.jar;C:\Users\VisitorDL1\.gradle\caches\modules-2\files-2.1\com.google.code.findbugs\jsr305\3.0.1\f7be08ec23c21485b9b5a1cf1654c2ec8c58168d\jsr305-3.0.1.jar;C:\Users\VisitorDL1\.gradle\caches\modules-2\files-2.1\org.bouncycastle\bcpkix-jdk15on\1.51\6c8c1f61bf27a09f9b1a8abc201523669bba9597\bcpkix-jdk15on-1.51.jar;C:\Users\VisitorDL1\.gradle\caches\modules-2\files-2.1\com.jcraft\jzlib\1.1.3\c01428efa717624f7aabf4df319939dda9646b2d\jzlib-1.1.3.jar;C:\Users\VisitorDL1\.gradle\caches\modules-2\files-2.1\net.vrallev.ecc\ecc-25519-java\1.0.1\3d9b19747b4c140f202d0a49be2973c8f5ca3ab5\ecc-25519-java-1.0.1.jar;C:\Users\VisitorDL1\.gradle\caches\modules-2\files-2.1\org.slf4j\slf4j-api\1.7.7\2b8019b6249bb05d81d3a3094e468753e2b21311\slf4j-api-1.7.7.jar;C:\Users\VisitorDL1\.gradle\caches\modules-2\files-2.1\org.bouncycastle\bcprov-jdk15on\1.51\9ab8afcc2842d5ef06eb775a0a2b12783b99aa
Launcher Type: SUN_STANDARD

Environment Variables:
JAVA_HOME=C:\Program Files\Java\jdk1.8.0_92
CLASSPATH=C:\Users\VisitorDL1\Downloads\GRIP-master\GRIP-master\\gradle\wrapper\gradle-wrapper.jar
PATH=C:\ProgramData\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files (x86)\nodejs\;C:\Program Files (x86)\Microsoft SQL Server\Client SDK\ODBC\130\Tools\Binn\;C:\Program Files (x86)\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files (x86)\Microsoft SQL Server\130\DTS\Binn\;C:\Program Files (x86)\Microsoft SQL Server\130\Tools\Binn\ManagementStudio\;C:\Program Files\Microsoft\Web Platform Installer\;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin;C:\Program Files (x86)\GtkSharp\2.12\bin;C:\Program Files (x86)\Microsoft SQL Server\110\DTS\Binn\;C:\Program Files (x86)\Microsoft SQL Server\120\DTS\Binn\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Microsoft Emulator Manager\1.0\;C:\Program Files\Git\cmd;C:\Program Files\dotnet\;C:\dotnetgmrtool\Windows;C:\Users\VisitorDL1\AppData\Local\Microsoft\WindowsApps;
USERNAME=VisitorDL1
OS=Windows_NT
PROCESSOR_IDENTIFIER=Intel64 Family 6 Model 42 Stepping 7, GenuineIntel

---------------  S Y S T E M  ---------------

OS: Windows 10.0 , 64 bit Build 15063 (10.0.15063.296)

CPU:total 8 (4 cores per cpu, 2 threads per core) family 6 model 42 stepping 7, cmov, cx8, fxsr, mmx, sse, sse2, sse3, ssse3, sse4.1, sse4.2, popcnt, avx, aes, clmul, ht, tsc, tscinvbit

Memory: 4k page, physical 8344588k(3493596k free), swap 9655308k(3310780k free)

vm_info: Java HotSpot(TM) 64-Bit Server VM (25.92-b14) for windows-amd64 JRE (1.8.0_92-b14), built on Mar 31 2016 21:03:04 by "java_re" with MS VC++ 10.0 (VS2010)

time: Fri Aug 18 13:38:15 2017
elapsed time: 263 seconds (0d 0h 4m 23s)

```

JLLeitschuh, 2017-08-19:
Looks like you segfaulted somewhere running native code. What were you doing?

ThadHouse, 2019-10-29:
I think this is solved by #952. I've been testing for over an hour, where I would see this failure, and no longer see it. I think it was a bug in the old IP Frame Grabber code.

closed 2019-10-30 by AustinShalit