WPIRoboticsProjects/GRIP #863: Add ZeroMQ Support for output

marshallmassengill, 2017-09-22:
Would love to see ZeroMQ supported as an output.  It's a great way to manage messages and it is what we've been recommending to teams recently.  Not perfect but it works well for a lot of use cases.  Let us know how we can help!

JLLeitschuh, 2017-09-22:
Would we need to provide message definitions to teams so they could generate messages?
What format is the data sent?

marshallmassengill, 2017-09-22:
You can send just about anything.  We typically are sending values for coordinates.  I think you would need to provide some kind of information about the definition or type.  Here's a decent intro: https://www.digitalocean.com/community/tutorials/how-to-work-with-the-zeromq-messaging-library#zeromq-transport-types