WPIRoboticsProjects/GRIP #864: USB Camera settings (brightness, exposure etc)

jimakok, 2017-10-05:
Current GRIP setup does not allow the adjustment of USB camera settings.  As the distance to the target is increased, the camera automatically adjusts to ambient lighting conditions and therefore does not allow for the correct identification of target in many instances.

JLLeitschuh, 2017-10-05:
Do you want to be able to configure it as a feedback loop from the camera pipeline?

jimakok, 2017-10-05:
Is there a timetable for multi-platform support?

In the meantime, I would be fine with any alternative where I could set camera defaults prior to processing by GRIP.

JLLeitschuh, 2017-10-05:
@PeterJohnson?

jimakok, 2017-10-05:
ah.... I will see what I can learn about cscore there.

Yes, this is for FIRST.  USB camera is connected to Kangaroo where GRIP is run.

JLLeitschuh, 2017-10-05:
If you can run cscore on the Kangaroo you can connect  that to GRIP as a IP camera stream.

PeterJohnson, 2017-10-05:
Unfortunately cscore does not support USB cameras on Windows yet (which is what the Kangaroo is).  My ECD on this is early November.