WPIRoboticsProjects/GRIP #865: GRIP not deploying to Raspberry Pi 3

jcosx, 2017-10-22:
Hi everyone,

My team is planning to integrate vision processing this year for our robot but we keep on running into an issue in our terminal (pasted below). I am currently following the tutorial of the wiki and I have followed every step but this error is still present. I put both of the packages required in the correct directory and I have followed everything correctly 

Here is some stats for troubleshooting:
-current OS: Raspbian Jessie Lite, downloaded from here: https://howtoraspberrypi.com/downloads/ (we first downloaded it from the raspberry pi site, but it gave us SSH issues)
-Model: Raspberry Pi 3 Model B
-GRIP version: 1.5.2
-Debian version 9.1
-running oracle-java8-jdk package

Here is our error message:

`#`
`# A fatal error has been detected by the Java Runtime Environment:`
`#`
`#  SIGSEGV (0xb) at pc=0x766fd6f4, pid=565, tid=1983489136`
`#`
`# JRE version: Java(TM) SE Runtime Environment (8.0_65-b17) (build 1.8.0_65-b17)`
`# Java VM: Java HotSpot(TM) Client VM (25.65-b01 mixed mode linux-arm )`
`# Problematic frame:`
`# V  [libjvm.so+0x2e36f4]`
`#`
`# Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c `
`unlimited" before starting Java again`
`#`
`# An error report file with more information is saved as:`
`# /home/pi/hs_err_pid565.log`
`#`
`# If you would like to submit a bug report, please visit:`
`#   http://bugreport.java.com/bugreport/crash.jsp`
`#`
`Aborted`

If any more information is needed, please let me know. I've looked at previous posts, including issue #690 and #366. I have also taken a look at the WPIRoboticsProjects/GRIP gitter chat room but nothing has worked. I heard some people say that it's a bug in java, but if it is, is there a way around it? My team has hit a roadblock at this point since we don't know how to approach the issue, so any help would be great help to us.

Thanks!

JLLeitschuh, 2017-10-23:
We haven’t ever supported deployment to a PI officially.

You may want to look into the code generation feature which should work for running on a PI

AustinShalit, 2017-10-23:
I would take a look at using https://github.com/wpilibsuite/VisionBuildSamples/tree/master/Java combined with GRIP code generation.  I can write a more in depth tutorial later this week but that should get you started.

robovol, 2017-10-23:
Are you able to run a "hello world"  app on the pi?  Based on the fact you ran into an SSH issue using the official Raspbian might represent an issue on setting up the Raspberry pi.  You should be able to write a hello world(similar to this: http://www.instructables.com/id/Hello-world-by-Python-on-Raspberry-PI/).  Next step, check your video.  Are you able to get a webstream of your camera?  An improperly set up web cam can result in a crash.