WPIRoboticsProjects/GRIP #875: Input must have a value to run this step

Einfari24, 2017-12-28:
I'm new to using GRIP, but I keep having the same issue no matter what webcam I use, or if I run GRIP from the exe or directly build it from IntelliJ. If I open up an image to look at and then add any step, it gives me an error saying this step could not be run as "input must have a value to run this step." I can't find any answers on forums either. I am using Windows 8.1. Here is what the error looks like. 
[image]

SamCarlberg, 2017-12-28:
You need to specify the source for the input image. Click the circle on the webcam block labeled "Image", then drag it to the circle on the HSV block labeled "Input". All steps are connected in this way

closed 2017-12-28 by SamCarlberg

Einfari24, 2017-12-28:
Thank you! That was a stupid mistake of mine. Now I got it working.