WPIRoboticsProjects/GRIP #879: Feature Request: Unwrap 360 Camera

npetrangelo, 2018-01-18:
Is it possible to add an Unwrap360Camera operation to GRIP? I know OpenCV does it, so I think it should be possible. I would use it with a 360 camera on my team's robot, so we could see the target in any direction.

JLLeitschuh, 2018-01-18:
This would be the right place to ask for the feature request. Do you know the name of the method (or sample code from some documentation) that allows you to unwrap a 360 camera image?

gftabor, 2018-01-19:
Well "unwrap" isn't a standard term in opencv. Matlab has a lot of correction code for fisheye/360 lenses. 
[image]

https://www.mathworks.com/help/vision/ug/fisheye-calibration-basics.html?requestedDomain=true

Now I have some code I consider "unwrapping" to produce this image.

[image]

The code can be found here
https://github.com/asgard-mqp/warper/blob/master/src/warper_node.cpp

The code just remaps the image into new coordinate system, X axis is angle and Y axis is pixel radius. All based on the center of the image. This image is designed so that as you rotate in place and have the objects just slide sideways along the image.

npetrangelo, 2018-01-22:
Most 360 cameras are actually two back-to-back 180 fishere cameras, so I want a way to convert that to an equirectangular projection, assuming a 360 camera serves the two fisheye frames. The idea is if my team installs the 360 camera properly, we can use the x coordinate of a contour from a equirectangular projection as the angle almost directly. From there, it would be nice if the driver station camera stream could view it like Facebook's and YouTube's video players do it.
https://wiki.panotools.org/Equirectangular_Projection
P.S. Sorry if my wording is imprecise, I'm not always sure what the technical terms are.

JLLeitschuh, 2018-01-22:
It seems like something that could be implemented. Since you have a 360 camera you could test with would you be willing to try opening a PR with the operation?

npetrangelo, 2018-01-23:
I'm not sure if my team actually has a 360 camera at the moment, but I think one of the mentors does. I might be able to persuade him to let us use it for testing purposes. I'll get back to you on that.