WPIRoboticsProjects/GRIP #882: GRIP-SmartDashboard does not seem to work with latest version of GRIP. Is there a replacement?

gartaud, 2018-02-03:
Hi

We have happily used the GRIP-SmartDashboard plugin to display contours found by GRIP on the dashboard in the past, but we have no luck doing so this year.

It is not clear if the GRIP-SmartDashboard is still supported. How should one display contours found by GRIP on the dashboard this year?

Thanks.

JLLeitschuh, 2018-02-20:
@SamCarlberg or @AustinShalit do one of you want to look into this?

SamCarlberg, 2018-02-20:
I can't think of a reason why it wouldn't work. Have you tried using it this year?

AustinShalit, 2018-02-20:
Did we switch GRIP to use cscore for camera streaming?

SamCarlberg, 2018-02-20:
No, that PR is still open

gartaud, 2018-02-21:
We tried, but somehow it does not work. We were unsure it we should expect it to still work or not...