WPIRoboticsProjects/GRIP #883: Any documentation on how to add new OpenCV functions?

gunasekaran7, 2018-02-19:
I am planning to add other OpenCV functions like Feature detectors like ORB. Is there any documentation on things to look out for adding such features?

JLLeitschuh, 2018-02-20:
For very simple OpenCV function, you can have a look at the examples we have here:
https://github.com/WPIRoboticsProjects/GRIP/blob/master/core/src/main/java/edu/wpi/grip/core/operations/CVOperations.java

For functions with more complicated behaviour, have a look at these examples:
https://github.com/WPIRoboticsProjects/GRIP/tree/master/core/src/main/java/edu/wpi/grip/core/operations/composite

If you want a very long winded (1.5 hours) explanation of the GRIP infrastructure, I did this talk here to explain how all of GRIP works.
https://www.youtube.com/watch?v=9V7T5YxzBz0

gunasekaran7, 2018-02-21:
@JLLeitschuh Thank you! I will work on it  and get back to you.

JLLeitschuh, 2018-02-21:
Cheers!
Please consider contributing any operations you add back to this repository.