WPIRoboticsProjects/GRIP #889: Nothing is working

hayderma, 2018-06-07:
I appreciate that this project took lot of time and effort having built JavaFX applications myself, but I am sorry to say, this is as good as useless, I can't even generate code, all it says is what is shown in the picture below, no documentation to solve this issue, where do I even Start? I know no one likes documentation but your time would be wasted if the users can't even use it, I was very excited about this because from what I seen in the pictures it allows me to visually manipulate pictures with sliders rather than code which makes it easy for me to understand the effects of everything, I am depressed now, and kind of pissed too, I've been trying too long.

[image]

nightpool, 2018-06-07:
I would suggest that you try following the [Introduction To GRIP](https://wpilib.screenstepslive.com/s/4485/m/24194/l/463566-introduction-to-grip) tutorial, or peruse some of the other [documentation](https://wpilib.screenstepslive.com/s/currentCS/m/vision/c/194953)

SamCarlberg, 2018-06-07:
Most documenation is on [wpilib.screenstepslive.com](http://wpilib.screenstepslive.com/s/currentCS/m/vision/c/194953). I suggest reading the "Introduction to GRIP" and "Generating Code from GRIP" pages. You can easily open this page through GRIP from the `Help > About` dialog.

There is also some documentation on the [wiki](https://www.github.com/wpiroboticsprojects/grip/wiki), but screensteps is really where the bulk of the documentation is located.

npetrangelo, 2019-01-07:
That dialog says it's because there are no sources in the pipeline. To put it another way, what code do you expect that to generate? An empty file?

I suggest you add a source, an image would do. That's the minimum you need to export code.

closed 2019-01-08 by AustinShalit