WPIRoboticsProjects/GRIP #893: Python Sobel implementation ignores defined kernel size

UsaidPro, 2018-10-12:
Hello,
When I was using GRIP's Sobel edge detection, I found that the Mat returned from the Sobel I saw on GRIP did not match what the generated Python code created. Instead, the Mat would always have a ksize = 3 (the default value) no matter what kernel size I set.

This is what appeared in GRIP:
[image]

This is what appeared in Python:
[image]

The bottom image (the Sobel edge) exactly matches the Mat I see in GRIP when I use kernel size 3.
Laplacian doesn't have this issue however.

The solution I found:
Change this - 
```
def __cv_sobel(src, dx, dy, k_size, scale, delta, border_type):
      """Find edges by calculating the requested derivative order for the given image.
      Args:
         src: A numpy.ndarray.
         k_size: Odd number that is size of the kernel.
         scale: Scaling factor for Sobel.
         delta: Offset for values in Sobel.
         border_type: Opencv enum.
      Returns:
         The result as a numpy.ndarray.
      """
      return cv2.Sobel(src, 0, (int)(dx + 0.5), (int)(dy + 0.5), (int)(k_size + 0.5),
                     scale = scale, delta = delta, borderType = border_type)
```
to this.
```
def __cv_sobel(src, dx, dy, k_size, scale, delta, border_type):
      """Find edges by calculating the requested derivative order for the given image.
      Args:
         src: A numpy.ndarray.
         k_size: Odd number that is size of the kernel.
         scale: Scaling factor for Sobel.
         delta: Offset for values in Sobel.
         border_type: Opencv enum.
      Returns:
         The result as a numpy.ndarray.
      """
      return cv2.Sobel(src, 0, (int)(dx + 0.5), (int)(dy + 0.5), ksize=(int)(k_size + 0.5),
                     scale = scale, delta = delta, borderType = border_type)
```
Adding a "ksize=" to directly specify the ksize works (this is what the Laplacian function does). Directly specifying the ksize sets the kernel size correctly.

JLLeitschuh, 2018-10-12:
Would you be willing to open a PR to fix this issue?

UsaidPro, 2018-10-12:
Sure! I'll create one this afternoon.

JLLeitschuh, 2018-10-12:
Thanks! @SamCarlberg Any reason I couldn't kick out a release?

SamCarlberg, 2019-04-20:
Closed by #894

closed 2019-04-20 by SamCarlberg