WPIRoboticsProjects/GRIP #912: Should previews be interactive?

SamCarlberg, 2019-01-22:
Thinking this would be useful mostly for filtering operations where the parameters aren't obvious.

Like hovering over a previewed contour could show its bounding box properties (cx, cy, w, h), area, solidity, etc. 

@JLLeitschuh Thoughts?

JLLeitschuh, 2019-01-22:
Would this information also be available as an output from the operation or would it be purely visual in the preview?

SamCarlberg, 2019-01-22:
It would only preview the operation's output. The GIFs above use the width, height, and center point of the contour, all of which are included in the `ContoursReport` output of the operation

JLLeitschuh, 2019-01-22:
Okay, this sounds like a good idea.

SamCarlberg, 2019-01-22:
Dumping info into a tooltip:

[image]