WPIRoboticsProjects/GRIP #913: Video file sources loaded from save files do not have controls

SamCarlberg, 2019-01-22:
**Describe the bug**  
A video file source that is loaded from a save file does not have the play/pause button and scrubber controls.

**To Reproduce**  
Steps to reproduce the behavior:
1. Add a new video file source
2. Save the project
3. Load the saved project

**Expected behavior**  
Video files sources should always have controls

JLLeitschuh, 2019-01-22:
Have any idea what the root cause is? I don't think I ever touched the video playback element.

SamCarlberg, 2019-01-22:
The controller only checks if the controls should be added if there's multiple frames in the video file, since it also supports opening regular image files or video files with only a single frame. That check can probably just be removed.

The underlying issue is that a source created from the UI is initialized _before_ posting a `SourceAddedEvent`, but a source created when loading a save file is initialized _after_ posting the event. The UI element is created when the event is fired, checks the number of frames (which is -1 if the source is not yet initialized), then creates the controls only if the frame count is >= 2. So another solution would be to fire the `SourceAddedEvent` _after_ initializing the source when loading a save file

PR #914 (2019-01-28): Initialize sources loaded from save file before firing event
Fixes a problem with video file sources loaded from a save file not having play/pause and scrubber controls

Closes #913
closed 2019-01-28 by SamCarlberg