WPIRoboticsProjects/GRIP #920: generated java code for filterLines() is incorrect

mfkahn, 2019-02-03:
Version: latest GRIP Mac OSX Binary 1.5.2

When using Filter Lines, generated GripPipeline.java is incorrect:

```
@Override public void process(Mat source0) {
      ...
      filterLines(filterLinesLines, filterLinesMinLength, filterLinesAngle, filterLinesOutput);
}

...

	private void filterLines(List<Line> inputs,double minLength,double[] angle,
		List<Line> outputs) {
		outputs = inputs.stream()
				.filter(line -> line.lengthSquared() >= Math.pow(minLength,2))
				.filter(line -> (line.angle() >= angle[0] && line.angle() <= angle[1])
				|| (line.angle() + 180.0 >= angle[0] && line.angle() + 180.0 <= angle[1]))
				.collect(Collectors.toList());
	}
```

outputs is passed by reference from `process()`, simply reassigning `outputs` to a different List has no effect on the instance property filterLinesOutput, which is what needs to be filled with the detected lines.  The impact is line count in the GUI will seem to work, but the generated code will always have ZERO lines returned by `filterLinesOutput()`, and students will not know why their vision pipeline works in GRIP but not in their bot.

Potential corrections:

```
    private void filterLines(List<Line> inputs,double minLength,double[] angle,
		List<Line> outputs) {
                outputs.clear();
		outputs.addAll(inputs.stream()
				.filter(line -> line.lengthSquared() >= Math.pow(minLength,2))
				.filter(line -> (line.angle() >= angle[0] && line.angle() <= angle[1])
				|| (line.angle() + 180.0 >= angle[0] && line.angle() + 180.0 <= angle[1]))
				.collect(Collectors.toList()));
    }

```
or
```
@Override public void process(Mat source0) {
      ...
      filterLines(filterLinesLines, filterLinesMinLength, filterLinesAngle); // remove last arg
}
   
    private void filterLines(List<Line> inputs,double minLength,double[] angle) {
                
		filterLinesOutput = inputs.stream()
				.filter(line -> line.lengthSquared() >= Math.pow(minLength,2))
				.filter(line -> (line.angle() >= angle[0] && line.angle() <= angle[1])
				|| (line.angle() + 180.0 >= angle[0] && line.angle() + 180.0 <= angle[1]))
				.collect(Collectors.toList());
    }

```

JLLeitschuh, 2019-02-04:
Hi @mfkahn, thanks for opening the issue. Any chance you have a moment to open a PR to fix the issue?

mfkahn, 2019-02-04:
Sure - I think I will go with option #1, as it is consistent with how findLines is implemented.  I should have time to submit a PR in the next couple of days.

commit d8cabe83 (2019-02-10): Fixes #920 incorrect java codegen for filter lines (#923)

Original VM template was reassigning the filtered list to the outputs variable
which would have no effect on the contents of the passed-in argument.  Working
pipelines in the GRIP GUI would show zero filtered lines when the generated class
was used.
closed 2019-02-10 by SamCarlberg