WPIRoboticsProjects/GRIP #924: Rename Debian Package Name

xDangerousDinox, 2019-02-24:
On Debian based linux the package manager sees grip as another package also name grip(some github markdown program). I believe renaming this to something like grip-vision should fix the problem of upgrading the package manager to delete current version of this grip and update to the other grip package.

JLLeitschuh, 2019-02-24:
Is this version of GRIP available on the Debian package manager? I was unaware that it had been published there.

xDangerousDinox, 2019-02-24:
[image]

There is a duplicate grip package in the Debian package manager. I installed GRIP with sudo dpkg -i GRIP on the available .deb file. When I sudo apt upgrade the system it replaces GRIP with the grip on the Debian Repositories. 

[image]

On the image above you can see it is removing 300 MB of disk space to install the grip on the Debian repositories.

JLLeitschuh, 2019-02-24:
@AustinShalit Do you know anything about this? I'd recommend only installing GRIP from our official release page.

https://github.com/WPIRoboticsProjects/GRIP/releases

xDangerousDinox, 2019-02-24:
I did use the installer from official release page. However, the package manager confuses our GRIP with this grip https://tracker.debian.org/pkg/grip