WPIRoboticsProjects/GRIP #958: Not Opening File

MikaO13, 2020-01-16:
I was working on a project to establish a pipeline for a limelight on Saturday, but when I tried to reopen the same file on Monday, it wouldn't work. Also, all of the image processing options are gone. I restarted my computer, and also uninstalled and reinstalled GRIP, but it just isn't working :/

JLLeitschuh, 2020-01-16:
Can you upload a copy of your file?

MikaO13, 2020-01-16:
[findyellow.zip](https://github.com/WPIRoboticsProjects/GRIP/files/4068977/findyellow.zip)
Here, thank you for responding so fast

JLLeitschuh, 2020-01-16:
This is the error I get when opening your file.

```
Thread[Project Open Thread,5,main] threw an exception
## Stack Trace:
```java
com.thoughtworks.xstream.converters.ConversionException: 
---- Debugging information ----
cause-exception     : java.lang.IllegalArgumentException
cause-message       : null
class               : edu.wpi.grip.core.settings.CodeGenerationSettings
required-type       : edu.wpi.grip.core.settings.CodeGenerationSettings
converter-type      : edu.wpi.grip.core.serialization.CodeGenerationSettingsConverter
line number         : 95
class[1]            : edu.wpi.grip.core.Pipeline
converter-type[1]   : com.thoughtworks.xstream.converters.reflection.ReflectionConverter
version             : v1.5.2
-------------------------------
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:79)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66)
	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.unmarshallField(AbstractReflectionConverter.java:503)
	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.doUnmarshal(AbstractReflectionConverter.java:429)
	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.unmarshal(AbstractReflectionConverter.java:281)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:50)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.start(TreeUnmarshaller.java:134)
	at com.thoughtworks.xstream.core.AbstractTreeMarshallingStrategy.unmarshal(AbstractTreeMarshallingStrategy.java:32)
	at com.thoughtworks.xstream.XStream.unmarshal(XStream.java:1230)
	at com.thoughtworks.xstream.XStream.unmarshal(XStream.java:1214)
	at com.thoughtworks.xstream.XStream.fromXML(XStream.java:1085)
	at edu.wpi.grip.core.serialization.Project.open(Project.java:122)
	at edu.wpi.grip.core.serialization.Project.open(Project.java:101)
	at edu.wpi.grip.ui.MainWindowController.lambda$openProject$5(MainWindowController.java:197)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.IllegalArgumentException
	at com.google.common.base.Preconditions.checkArgument(Preconditions.java:108)
	at edu.wpi.grip.core.settings.CodeGenerationSettings$Builder.language(CodeGenerationSettings.java:148)
	at edu.wpi.grip.core.settings.CodeGenerationSettings.builder(CodeGenerationSettings.java:124)
	at edu.wpi.grip.core.serialization.CodeGenerationSettingsConverter.unmarshal(CodeGenerationSettingsConverter.java:40)
	at edu.wpi.grip.core.serialization.CodeGenerationSettingsConverter.unmarshal(CodeGenerationSettingsConverter.java:19)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72)
	... 16 more

```
## System Info:

Property Name | Property
 ----- | -----
GRIP Version | v1.5.2
java.version | 1.8.0_92
javafx.version | 8.0.92
os.name | Mac OS X
os.version | 10.15.2
os.arch | x86_64
```

MikaO13, 2020-01-16:
That's the error I get too. Is it with the file then and not GRIP?
But if it's with the file, I still can't get the image processing things on GRIP like blur.

JLLeitschuh, 2020-01-16:
Here's a fixed version of your save file.
```xml
<grip:Pipeline>
  <sources>
    <grip:Camera>
      <property name="deviceNumber" value="0"/>
    </grip:Camera>
  </sources>
  <steps>
    <grip:Step name="Blur">
      <grip:Input step="0" socket="0"/>
      <grip:Input step="0" socket="1">
        <value>BOX</value>
      </grip:Input>
      <grip:Input step="0" socket="2">
        <value>2.2522522522522523</value>
      </grip:Input>
      <grip:Output step="0" socket="0" previewed="false"/>
    </grip:Step>
    <grip:Step name="HSL Threshold">
      <grip:Input step="1" socket="0"/>
      <grip:Input step="1" socket="1">
        <value>
          <double>24.28057553956835</double>
          <double>70.95563139931741</double>
        </value>
      </grip:Input>
      <grip:Input step="1" socket="2">
        <value>
          <double>113.51169064748201</double>
          <double>255.0</double>
        </value>
      </grip:Input>
      <grip:Input step="1" socket="3">
        <value>
          <double>52.74280575539568</double>
          <double>149.47525597269626</double>
        </value>
      </grip:Input>
      <grip:Output step="1" socket="0" previewed="true"/>
    </grip:Step>
    <grip:Step name="CV erode">
      <grip:Input step="2" socket="0"/>
      <grip:Input step="2" socket="1"/>
      <grip:Input step="2" socket="2"/>
      <grip:Input step="2" socket="3">
        <value>1</value>
      </grip:Input>
      <grip:Input step="2" socket="4">
        <value>BORDER_CONSTANT</value>
      </grip:Input>
      <grip:Input step="2" socket="5"/>
      <grip:Output step="2" socket="0" previewed="true"/>
    </grip:Step>
    <grip:Step name="Find Contours">
      <grip:Input step="3" socket="0"/>
      <grip:Input step="3" socket="1">
        <value>false</value>
      </grip:Input>
      <grip:Output step="3" socket="0" previewed="true"/>
    </grip:Step>
  </steps>
  <connections>
    <grip:Connection>
      <grip:Output step="2" socket="0" previewed="true"/>
      <grip:Input step="3" socket="0"/>
    </grip:Connection>
    <grip:Connection>
      <grip:Output step="0" socket="0" previewed="false"/>
      <grip:Input step="1" socket="0"/>
    </grip:Connection>
    <grip:Connection>
      <grip:Output step="1" socket="0" previewed="true"/>
      <grip:Input step="2" socket="0"/>
    </grip:Connection>
    <grip:Connection>
      <grip:Output source="0" socket="0" previewed="true"/>
      <grip:Input step="0" socket="0"/>
    </grip:Connection>
  </connections>
  <settings>
    <teamNumber>0</teamNumber>
    <publishAddress>roboRIO-0-FRC.local</publishAddress>
    <deployAddress>roboRIO-0-FRC.local</deployAddress>
    <deployDir>/home/lvuser</deployDir>
    <deployUser>lvuser</deployUser>
    <deployJavaHome>/usr/local/frc/JRE/</deployJavaHome>
    <deployJvmOptions>-Xmx50m -XX:-OmitStackTraceInFastThrow -XX:+HeapDumpOnOutOfMemoryError -XX:MaxNewSize=16m</deployJvmOptions>
  </settings>
  <codeGenerationSettings>
    <language>Java</language>
    <className>ReadYellow</className>
    <implementWpilibPipeline>false</implementWpilibPipeline>
    <saveDir>C:\Users\Mika\Desktop</saveDir>
    <packageName></packageName>
    <moduleName>grip</moduleName>
  </codeGenerationSettings>
</grip:Pipeline>
```

[findyellow-fixed.grip.zip](https://github.com/WPIRoboticsProjects/GRIP/files/4069012/findyellow-fixed.grip.zip)

Somehow your `<language>Java</language>` ended up being `<language>limelight</language>` which GRIP has no idea how to handle. I have no idea how that ended up happening.

MikaO13, 2020-01-16:
It's still giving me a problem (same error message), I attached a picture of what GRIP looks like for me. 
[image]

JLLeitschuh, 2020-01-16:
Can you please use the `Copy to Clipboard` and paste the contents here in a report?

MikaO13, 2020-01-16:
Thread[Project Open Thread,5,main] threw an exception
## Stack Trace:
```java
com.thoughtworks.xstream.mapper.CannotResolveClassException: grip:Pipeline
	at com.thoughtworks.xstream.mapper.DefaultMapper.realClass(DefaultMapper.java:81)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:125)
	at com.thoughtworks.xstream.mapper.DynamicProxyMapper.realClass(DynamicProxyMapper.java:55)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:125)
	at com.thoughtworks.xstream.mapper.PackageAliasingMapper.realClass(PackageAliasingMapper.java:88)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:125)
	at com.thoughtworks.xstream.mapper.ClassAliasingMapper.realClass(ClassAliasingMapper.java:79)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:125)
	at com.thoughtworks.xstream.mapper.ArrayMapper.realClass(ArrayMapper.java:74)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:125)
	at com.thoughtworks.xstream.mapper.SecurityMapper.realClass(SecurityMapper.java:71)
	at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:125)
	at com.thoughtworks.xstream.mapper.CachingMapper.realClass(CachingMapper.java:47)
	at com.thoughtworks.xstream.core.util.HierarchicalStreams.readClassType(HierarchicalStreams.java:29)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.start(TreeUnmarshaller.java:133)
	at com.thoughtworks.xstream.core.AbstractTreeMarshallingStrategy.unmarshal(AbstractTreeMarshallingStrategy.java:32)
	at com.thoughtworks.xstream.XStream.unmarshal(XStream.java:1486)
	at com.thoughtworks.xstream.XStream.unmarshal(XStream.java:1466)
	at com.thoughtworks.xstream.XStream.fromXML(XStream.java:1337)
	at edu.wpi.grip.core.serialization.Project.open(Project.java:122)
	at edu.wpi.grip.core.serialization.Project.open(Project.java:101)
	at edu.wpi.grip.ui.MainWindowController.lambda$openProject$5(MainWindowController.java:197)
	at java.lang.Thread.run(Thread.java:748)

```
## System Info:

Property Name | Property
 ----- | -----
GRIP Version | v1.5.2-17-gc3c694e
java.version | 1.8.0_181
javafx.version | 8.0.181
os.name | Windows 10
os.version | 10.0
os.arch | amd64

JLLeitschuh, 2020-01-16:
I think @SamCarlberg fixed this. Unfortunately, we haven't had a release yet with the fix. If you checkout the project and run it from the command line, it may fix your issue.

@SamCarlberg Can we ship a release so that we fix this issue for people?

JLLeitschuh, 2020-01-16:
This issue was resolved already.
https://github.com/WPIRoboticsProjects/GRIP/issues/784

It looks like the version of GRIP that the Limelight guys built the version of GRIP they had you download doesn't have this fix yet.

I'd recommend bothering them about this.

I wish they'd reached out to me about implementing Limelight support for GRIP. I'd have been fine if they implemented it in our repository proper as long as they were willing to help maintain the project.

MikaO13, 2020-01-16:
Oh, ok thank you. Can I just download another version of GRIP that works, or do I need to ask my team?

JLLeitschuh, 2020-01-16:
We haven't shipped an official release. You'd have to check out the code from source and build it. Limelight looks like they have a custom version of GRIP they built for you though. You'll have to send them a link to this issue.

MikaO13, 2020-01-16:
Ok, thank you so much for your help!

JLLeitschuh, 2020-01-16:
I'm going to leave this open to track our conversation with the Limelight guys

MikaO13, 2020-01-16:
Ok.
If it helps, this is what I downloaded (project leader told us all to download it)
https://www.mediafire.com/file/w7dphr1y46mw0az/Grip_Limelight.zip/file

JLLeitschuh, 2020-01-16:
Yea, I found it. Thanks!