WPIRoboticsProjects/GRIP #959: Webcam Not Working on Mac

Draeorc, 2020-01-23:
After successfully launching Grip, I tried to add a webcam source in the 0 slot. I also tried the other slots but they didn't work either.

Thread[JavaFX Application Thread,5,main] threw an exception
## Stack Trace:
```java
java.lang.IllegalStateException: Can not start while in state STARTING. Must be in one of states: NEW, FAILED, TERMINATED
	at edu.wpi.grip.core.util.service.AutoRestartingService.startAsync(AutoRestartingService.java:123)
	at edu.wpi.grip.core.util.service.AutoRestartingService.startAsync(AutoRestartingService.java:108)
	at edu.wpi.grip.core.sources.CameraSource.startAsync(CameraSource.java:270)
	at edu.wpi.grip.core.sources.CameraSource.startAsync(CameraSource.java:46)
	at edu.wpi.grip.ui.components.StartStoppableButton.lambda$new$1(StartStoppableButton.java:64)
	at com.sun.javafx.event.CompositeEventHandler$NormalEventFilterRecord.handleCapturingEvent(CompositeEventHandler.java:282)
	at com.sun.javafx.event.CompositeEventHandler.dispatchCapturingEvent(CompositeEventHandler.java:98)
	at com.sun.javafx.event.EventHandlerManager.dispatchCapturingEvent(EventHandlerManager.java:223)
	at com.sun.javafx.event.EventHandlerManager.dispatchCapturingEvent(EventHandlerManager.java:180)
	at com.sun.javafx.event.CompositeEventDispatcher.dispatchCapturingEvent(CompositeEventDispatcher.java:43)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:52)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.EventUtil.fireEventImpl(EventUtil.java:74)
	at com.sun.javafx.event.EventUtil.fireEvent(EventUtil.java:54)
	at javafx.event.Event.fireEvent(Event.java:198)
	at javafx.scene.Scene$MouseHandler.process(Scene.java:3757)
	at javafx.scene.Scene$MouseHandler.access$1500(Scene.java:3485)
	at javafx.scene.Scene.impl_processMouseEvent(Scene.java:1762)
	at javafx.scene.Scene$ScenePeerListener.mouseEvent(Scene.java:2494)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(GlassViewEventHandler.java:380)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(GlassViewEventHandler.java:294)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler.lambda$handleMouseEvent$354(GlassViewEventHandler.java:416)
	at com.sun.javafx.tk.quantum.QuantumToolkit.runWithoutRenderLock(QuantumToolkit.java:389)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler.handleMouseEvent(GlassViewEventHandler.java:415)
	at com.sun.glass.ui.View.handleMouseEvent(View.java:555)
	at com.sun.glass.ui.View.notifyMouse(View.java:937)

```
## System Info:

Property Name | Property
 ----- | -----
GRIP Version | v1.5.2
java.version | 1.8.0_92
javafx.version | 8.0.92
os.name | Mac OS X
os.version | 10.15.2
os.arch | x86_64

JLLeitschuh, 2020-02-13:
Can you try the version on master?

srajanpm, 2021-03-25:
I am experiencing a similar issue to with v 1.5.2 

edu.wpi.grip.core.sources.GrabberService$GrabberServiceException: [Webcam 1] Failed to start
	at edu.wpi.grip.core.sources.GrabberService.startUp(GrabberService.java:54)
	at com.google.common.util.concurrent.AbstractExecutionThreadService$1$2.run(AbstractExecutionThreadService.java:60)
	at com.google.common.util.concurrent.Callables$4.run(Callables.java:122)
	at java.lang.Thread.run(Thread.java:745)
Caused by: org.bytedeco.javacv.FrameGrabber$Exception: cvCreateCameraCapture() Error: Could not create camera capture.
	at org.bytedeco.javacv.OpenCVFrameGrabber.start(OpenCVFrameGrabber.java:179)
	at edu.wpi.grip.core.sources.GrabberService.startUp(GrabberService.java:52)
	... 3 more

JLLeitschuh, 2021-04-08:
@srajanpm can you also try running the code from source and see if that helps?