WPIRoboticsProjects/GRIP #961: LineSegmentDetector needs to be removed for license

sciencewhiz, 2020-02-05:
Per https://www.chiefdelphi.com/t/cpp-multicameraserver-wont-build/375965/7, GRIP generates code that uses LineSegmentDetector. Per https://github.com/opencv/opencv_contrib/issues/2016, LineSegmentDetector used Affero GPL v3 code and was removed. GRIP needs to remove the calls to LineSegmentDetector or relicense to a compatible license.

virtuald, 2020-02-05:
If GRIP generates the code but doesn't use it, doesn't that mean the end user's code is AGPLv3, not GRIP?

... but still, removing it would be a good idea.

sciencewhiz, 2020-02-05:
It's also used in GRIP to show the results. https://github.com/WPIRoboticsProjects/GRIP/blob/34afc3dfb07cb4327b079c4662e4581b044f0b0a/core/src/main/java/edu/wpi/grip/core/operations/composite/FindLinesOperation.java#L72

JLLeitschuh, 2020-02-06:
If it's incompatible with BSD, we need to remove our usage too.
CC: @SamCarlberg @AustinShalit

JLLeitschuh, 2020-09-08:
More information here:
https://github.com/opencv/opencv_contrib/issues/2524

ThadHouse, 2021-02-07:
https://github.com/opencv/opencv_contrib/issues/2016

Based on the latest post, it seems like the author realized the license issue was a mistake, and relicensed the code. So anything shipping with 3.x should be safe.

GrantPerkins, 2021-07-26:
https://github.com/WPIRoboticsProjects/GRIP/pull/976 migrates to FastLineDetector from LineSegmentDetector