WPIRoboticsProjects/GRIP #984: Microsoft HD-5000 webcam not working on GRIP (macOS)

neecola, 2022-01-19:
Hello there,
I'm working with GRIP on a MacBook Pro 2012 running Catalina, and it looks like the program doesn't recognize neither the Microsoft HD-5000 I plugged in via USB nor the built-in camera. I already checked with other programs on the same pc and both cameras work. 
GRIP says "Camera service error", and the error message is
```
du.wpi.grip.core.sources.GrabberService$GrabberServiceException: [Webcam 1] Failed to start
	at edu.wpi.grip.core.sources.GrabberService.startUp(GrabberService.java:54)
	at com.google.common.util.concurrent.AbstractExecutionThreadService$1$2.run(AbstractExecutionThreadService.java:60)
	at com.google.common.util.concurrent.Callables$4.run(Callables.java:122)
	at java.lang.Thread.run(Thread.java:745)
```

JLLeitschuh, 2022-01-20:
From the internal conversation about this from other maintainers:

> No, not surprised.  GRIP uses raw OpenCV, but cscore doesn't have USB camera support on mac because it's really hard/annoying to do.  I don't think opencv works either
> Maybe it works in opencv4, but grip is still using opencv3 

Personally, I thought this worked, but clearly something has changed since I originally wrote it. I don't have the time/resources to debug this issue, but I welcome outside contributions that solve this problem.

JLLeitschuh, 2022-01-20:
Here's where we create the OpenCV camera grabber.

https://github.com/WPIRoboticsProjects/GRIP/blob/2d30a10ced329c05ea55f675b609c6f9b6ee01e2/core/src/main/java/edu/wpi/grip/core/sources/CameraSource.java#L401

When using GRIP, the camera `0` should be the built-in camera. USB cameras should be higher numbers, usually 1, 2, 3, ect..

One thing you may want to check is that the MacOS permissions require that you grant camera access on a per-application basis. You may want to make sure you've given GRIP the permissions to access the camera.

team3572, 2023-02-07:
Just leaving this comment here for posterity. I know this isn't maintained anymore and don't expect this to be maintained.
Running on a M1 Mac on 12.6.2
GRIP doesn't even request permission to access the camera. And you can't manually add an app permission to the camera system, it has to be asked for.
I'm getting the same error as the OP.
At this point I would say the latest packaged release of GRIP is no longer viable on Macintosh.