WPIRoboticsProjects/GRIP #989: Focusing On Other Application While Saving Pipeline Crashes Grip on MacOS (Does Not Save Pipeline)

Braykoff, 2022-08-17:
When in GRIP on macOS, when you are saving the pipeline and choosing the save location, clicking out of the application (into any other application, or the desktop) causes GRIP to crash with no error, destroying any unsaved work.

[Video](https://drive.google.com/file/d/1cw37AL3sDXtpgOvia13zz7gltxpMu5E0/view?usp=sharing)

**Steps to Reproduce:**
- Open Grip
- Press File > Save
- Click into any other application (remove focus from GRIP)
- When the 'Choose Save Location' Finder window opens on GRIP, click into it
- Click into any other application again (remove focus from GRIP)
- GRIP crashes

I can not confirm if this happens on Windows too.

JLLeitschuh, 2022-08-18:
Do you have an exception from the log file you can share with us?

Braykoff, 2022-08-18:
@JLLeitschuh The log file doesn't have an exception:
```
Aug 18, 2022 1:00:41 PM edu.wpi.grip.core.util.service.LoggingListener starting
INFO: [PipelineRunner] Starting
Aug 18, 2022 1:00:47 PM edu.wpi.grip.core.events.EventLogger eventPosted
INFO: Event on thread 'JavaFX Application Thread': StepAddedEvent{step=Step{operation=Desaturate}, index=OptionalInt[0]}
Aug 18, 2022 1:00:47 PM edu.wpi.grip.core.events.EventLogger eventPosted
INFO: Event on thread 'JavaFX Application Thread': StepAddedEvent{step=Step{operation=Distance Transform}, index=OptionalInt[1]}
```
(I added random steps just to see if they would appear in the log)
When it crashes, nothing is added.
`GRIP.log.lck` is blank.

JLLeitschuh, 2022-08-18:
I have a sneaking suspicion this is out of our control to fix. It sounds like an OS bug

Braykoff, 2022-08-19:
@JLLeitschuh 

I can confirm that this does not occur in other macOS applications. I have tried the steps in Text Edit (the built-in text editor) and IDLE (Python IDE, neither built-in nor made by Apple). Neither application crashed when the 'save as' window was unfocused. This might be an OS bug, however, it only seems to affect GRIP.

SamCarlberg, 2022-08-29:
@Braykoff:

1. What version of GRIP are you experiencing this with?
2. What version of maOS?
3. What Apple hardware (eg Macbook Pro 2019)?

SamCarlberg, 2022-08-29:
FWIW, I can't repro this with a 2020 M1 Macbook Air